pageauc/pi-timolo #98: zoom in on area for motion detection

blokdk, 2018-07-19:
Is there a (easy) way to adapt the region for motion detection. All I need is detecting some change on a tree, vertical area, covering maybe a tenth of the full image. I noticed, I think in your speed camera project, you sort of specify the detection area. 
How could that be done in het pi-timolo project?

Thanks in advance,

pablok

pageauc, 2018-07-19:
Yes that can be done by cropping an area from the stream image and passing it to opencv to process. Track trigger length would need to be adjusted or could be just set as a percent of the crop image width or height.  
I have published pi-timolo.py release 10.99 that will allow showing the stream image detection area overlaid on the full size image.  IMPORTANT.  If you are using browser use scroll bars to see rest of image since full image will usually be bigger than the browser window.   I have reduce the motion tracking stream size to 320x240 (hard coded in pi-timilo.py line 206, 207.  It was previously 640x240.

To upgrade run menubox.sh UPGRADE menu pick.  It will upgrade to the latest pi-timolo.py.  Your existing config.py file will not be changed. config.py.new will have a new variable called  
 
    imageShowStream = False.  

You can simply add this new variable to your existing config.py and Change to True to display a box line area on image so you can align camera then change back to False when camera alignment is complete.  In future I will take a look at cropping area out of stream image for motion tracking but this was an easy alternative. 
 
If you want to return config.py back to default settings you can do the following commands in terminal session

    cd ~/pi-timolo
    cp config.py config.py.bak
    cp config.py.new config.py

You would then need to transfer settings from the bak file to the new config.py file.  Just as easy to add the new variable to your existing config..py file.  Change the variable to imageShowStream = True to show lines on the full size image.  I would temporarily turn on timelapse to get a full image without needing to activate motion.  To set timelapse and take image every 30 seconds edit config.py and change the following variables 

    timelapseOn = True
    timelapseTimer = 30

Use the web browser and look in recent/timelapse folder for latest images.  After alignment is complete you can set

    timelapseOn = False

Let me know your progress
Claude ...

pageauc, 2018-07-19:
Note published release 11.00 to added stream box logic to night shots as well as day.

blokdk, 2018-07-20:
Dear Mr. Pageau,
Amazing. Have not yet tested it, played with it, but so grateful with this mail. Thank you so much.
pablok

blokdk, 2018-07-22:
Dear mr. Pageau,

That was an interesting experience with the detection rectangle you provided. It steered me into your code trying to figure out what these variables are used for and where. So I changed them into something I would name the right third of the image and was not at all content with the motion pictures it provided. Later I found the 1.5 factor for the bigImage and understood why I ended up with an image 600 px wide and 900 high where my camera_width and camera_height were 400 by 600.
I really enjoyed learning something this way. Now I have commented out the quick time images that you take out of the scream; because that is faster. I decided against it and am taking the switching out of the stream for granted and use the full image.

Thank you,

pablok

closed 2018-10-18 by pageauc