pageauc/pi-timolo #99: Run a script on motion detection

aledthom, 2018-10-17:
Hi, I have pi-timolo up and running on a pi3 and it works great, thanks for all your work on it.

For the project I'm working on I need to run a python script when a motion detection event occurs. Also I need to be able to stop the motion detection while the script runs and restart it once the script has done what it needs to do (I believe this should already be possible simply by calling "pi-timolo.sh stop" and "pi-timolo.sh start"

While I could edit the source to do what I want, this would be overwritten if we update to any later version, would it be possible to add maybe a line in config for a script to be run when motion is detected? Thanks.

pageauc, 2018-10-17:
Yes.  I could add a variable to point to a file path or hardcode a
specified filename since it would need to be in the same directory as
pi-timolo.py to be imported otherwise there would need to be an __init__.py
file with the file to be imported.  If the file is found it would import
the code from the file and execute a function with a specific name.  The
specified function would execute when motion is found.  As you mentioned
this would avoid inserting code into the source and loosing changes on
update.
There is a function in the source called userMotionCodeHere() but that code
would be gone after an update.

My preference would be to hard code a file name called user-motion-code.py
that gets imported with pi-timolo startup
I would have a template file that already exists. File would contain a
function called userMotionCodeHere() where you
could put your code.  Initially function would do nothing except maybe do a
dummy print statement.
I would eliminate the existing userMotionCodeHere() function in
pi-timolo.py since it would be imported from the
external import file  user-motion-code.py
This should solve your problem and avoids adding another variable to the
config.py

Let me know your thoughts.
Claude ....

Claude ...

pageauc, 2018-10-18:
I have added new feature to load user motion code from a user_motion_code.py file.  This will be downloaded if it does not exist otherwise existing file will not be overwritten.  See comments.  Let me know if this resolves your problem.   Note if specialized python libraries need to be loaded you should add to pi-timolo initialization code.  Let me know if this is a problem and I can take a look at an easier way to import these possibly with an __init__.py file.

aledthom, 2018-10-18:
That's awesome thanks, I'll test it now.

closed 2019-02-05 by pageauc