pageauc/pi-timolo #100: Capturing Vehicle movements

raspberryrippl3, 2019-01-16:
Im unclear as to what config variable needs adjusting in order to capture moving traffic. At the moment if a vehicle enters the stream box (adjusted to 960 x 225) the video or still doesnt capture until the vehicle is half way out of the stream box

pageauc, 2019-01-16:
track_len_trig = 70 
This is the default distance in pixels that the object is tracked before taking the speed image. 
You could try increasing or decreasing this value to control position where final speed image is taken. 
If you are still having problems send me a copy of your config.py  and any associated plugin file if used.

Claude ...

pageauc, 2019-01-17:
Lock up is usually related to 
1 a weak power supply can definitely cause issues.  rpi3 needs 2.5 amps and hub may not be reliably supplying that,  In the past Poor power supply has caused me problems with Kodi and changing to better quality has resolved.
2  issue with SD card.  I have experienced this several times with failed SD cards or hard drive.  Power supply issues can also corrupt drives.
3  high temperature issues > 85 C.  CPU will throttle speed to reduce temp but if RPI is in a hot place or poor cooling like in a closed container this can cause problems.  menubox.sh has menu pick to show temperature.  
4 Camera driver issue.  In the past there were issue with camera library updates and older firmware.  This should not be as much of an issue with Stretch.  Usually in these cases the RPI would freeze and need a hard reboot.  Solution was to update firmware and do apt-get update and upgrade to resolve issue.  I have not seen this with recent raspbian builds like Stretch.

I have had several RPI-2's single core's running pi-timolo for several years at a time without problems.  One thing you have to be careful of is moving the current image/video file while it is still being written to.  That is why I developed the mvleavelast.sh script https://github.com/pageauc/pi-timolo/blob/master/source/mvleavelast.sh

I would recommend you first try a good quality power supply 2.5 amps or greater instead of hub. 

Claude ...

raspberryrippl3, 2019-01-17:
Thank you. I'm with you on the power supply and maybe SD card. I'll try an official PI supply and a new SD card. The Anker specified output is rated at 5V / 4.8A

Temperature isn't an issue as it's about 3 degrees C here at the moment, and I did an rpi-update etc. Thank you again

closed 2019-02-05 by pageauc