pageauc/pi-timolo #102: Auto Backup

logicaprogrammabile, 2019-03-17:
Hi
thank you for have developed this amazing program.

I have a question, is possible to automatic backup timelapse files when i attached usb drive to the raspberry pi? 

In my case i have a remote raspberry pi zero where no have internet connection. i would go one day at week where i have installed pizero 
and manually backup the files only attaching the usb drive. 
Is possible that?

regards
Marco

pageauc, 2019-03-17:
I often attach an external usb disk drive or sometimes a usb memory stick.  Please note that the external drives will need to be formatted as NTFS or a Unix Format like EXT4 since pi-timolo uses unix symlinks for the recent folder.
There is more information at the wiki but it is a little out of date and I should update it to be a bit easier to use.

https://github.com/pageauc/pi-timolo/wiki/How-to-mount-external-storage-drive

For my long duration timelapse projects I configure pi-timolo to save files in numerical sequence since clocks and things like daylight saving time can mess up the file sort order.  Numerical sequence works better since it does not rely on clock time for the file naming.
I also save all files to an external USB disk drive (old one Terabyte drive) that is formatted as ext4 unix format but NTFS format also works OK if you want to plug into a windows PC to transfer files.  You could also save to a NTFS or ext4 formatted USB stick but would have less storage space

One project takes a 1080p timelapse image every minute.  I currently have several years of images on this external drive.  When I return to the camera I connect via a network connection and transfer images for processing into timelapse videos.  That is what I do for my winter timelapse videos.

Note if I get some time I will update the wiki section on external drives but the current one should be usable as long as you format external drive as NTFS or a unix disk format.

Let me know if you need more help.
Regards Claude ...

pageauc, 2019-03-17:
Please note there may be some usbmount issues with Raspbian Stretch per this https://github.com/rbrito/usbmount/issues/2

Let me know if you plan to mount a USB disk drive or USB memory stick for external storage.
I can help you through setting up pi-timolo to use external storage.
Claude ...

logicaprogrammabile, 2019-03-17:
ok thank you, soon i try it this function

closed 2019-06-03 by pageauc