pageauc/pi-timolo #103: Trigger an event... launch an other py script...

thierryc, 2019-06-17:
Hello,

I would like to trigger an event on a RGPIO to switch on a light … or other things. 
What is the best way to implement this feature? It could be a plugin? 

Thank you in advance for your advices.

Thank you Claude for this very smart project.
pi-timolo running smoothly 24/24 on my R.pi B.

thierryc, 2019-06-17:
I found this in pi-timolo.py.

1703:                 if motionFound or forceMotion:

I could add after this line a if conditions to call an other py script (path in config) ?
Thank you in advance for your advices.

pageauc, 2019-06-17:
There is a script file called  user_motion_code.py that by default does
nothing but will get executed whenever motion is detected.  You should be
able to add your code into the function and your code will get executed
whenever motion is detected.

see code starting near line 90 and also line 1005 of pi-timolo.py for the
code related to this implementation.  You can also pass parameter eg
filename in this function.  This implementation is not too difficult to
customize if required.  Also make sure the code you add to the function
does not take too long since it will delay returning to motion tracking.
Note this does not work with timelapse., Only Motion tracking.   Note you
may need to add initialization code or global variables in config.py or as
part of pi-timolo.py

Let me know how you make out.

Regards Claude ....

thierryc, 2019-06-17:
Smart and simple. Thank you for your advice. 

I will use user_motion_code.py.

I keep you posted.

Thank you 
Thierry

closed 2019-07-20 by pageauc