pageauc/pi-timolo #104: Datestamp on Timelapse images missing if motion set to video

koila91, 2019-07-19:
Hi, Thanks for the great sogtware.

I noticed that the timestamps on the timelapse images disappeared after I switched the motion to  video.

The following combination works as expected, each image gets "stamped" with date/time :
showDateOnImage = True
motionVideoOn = False

But the following combination removes the datestamp from timelapse images:
showDateOnImage = True
motionVideoOn = True
--> the date is shown on motion video but not in timelapse images

pageauc, 2019-07-19:
Thanks.

I will look at the logic, See why this is happening,  Fix and release update

Will let you know
Claude ...

pageauc, 2019-07-19:
I have updated pi-timolo.py to Release 13.1 that should fix date on image
issue.
I have tested and seems OK but would appreciate your testing results

To update run menubox.sh  UPGRADE menu pick.  Your existing config.py will
not be changed.

Your feedback is greatly appreciated.
Claude ...

pageauc, 2019-07-19:
Still not working.  Will let you know when I figure it out

pageauc, 2019-07-20:
OK  Release 11.32 seems to be working.  I had to add try: except when
writing data on images and the file was a video file
Originally this is why I excluded writing data on images if motion video
was on since post processing could also try to write data onto a video file
and would fail   Now I just trap the error and pass.  That way timelapse
and motion video will work correctly together.

Anyway give this version a try and let me know if you have any problems
Claude ....

koila91, 2019-07-20:
Hi, thanks for the update **- now it seems to work properly.** 

Now the timelapse video I make daily has the timestamps on each frame.

Thanks!

closed 2019-07-20 by koila91

pageauc, 2019-07-20:
Did review of code and improved logging messages for config.py variable
checks.  Also added save recent to motionTrackQuickPic
Have not found any problems with update changes.
Regards Claude ...

pageauc, 2019-07-20:
Current version is 11.4