pageauc/pi-timolo #106: Curl install script Aborts during i2c apt-get downloads.

Distortionist, 2020-10-21:
I had to manually install those packages so the install script continues its job without failing.

Distortionist, 2020-10-21:
It did it again with different packages, here is what happens :
The following additional packages will be installed:
  python3-smbus
The following NEW packages will be installed:
  python3-pantilthat python3-smbus
0 upgraded, 2 newly installed, 0 to remove and 17 not upgraded.
Need to get 18.7 kB of archives.
After this operation, 117 kB of additional disk space will be used.
Do you want to continue? [Y/n] Abort.

Obviously I don't tell it to abort.

pageauc, 2020-10-21:
Updated script to add missing -yq to some apt-get commands
Thanks for raising the issue.  Claude ....

closed 2020-10-21 by pageauc