pageauc/pi-timolo #109: Can't start pi-timolo

Distortionist, 2020-10-23:
I went through all the settings, and when I run ./pi-timolo start it tries and quit after a bit.

If I try to run ./pi-timolo.py I get this error :

2020-10-23 20:45:46 INFO     <module>   Start pi-timolo per /home/pi/pi-timoloonfig.py Settings
Traceback (most recent call last):
  File "./pi-timolo.py", line 2425, in <module>
    timolo()
  File "./pi-timolo.py", line 1818, in timolo
    TIMELAPSE_DIR, TIMELAPSE_PREFIX)
  File "./pi-timolo.py", line 898, in subDirChecks
    if subDirCheckMaxHrs(subDirPath, maxHours, prefix):
  File "./pi-timolo.py", line 872, in subDirCheckMaxHrs
    dirDate = datetime.datetime.strptime(dirStr, "%Y-%m-%d-%H:%M")
  File "/usr/lib/python2.7/_strptime.py", line 332, in _strptime
    (data_string, format))
ValueError: time data '2020-1023-2028' does not match format '%Y-%m-%d-%H:%M'

pageauc, 2020-10-24:
> 2020-10-23 20:45:46 INFO Start pi-timolo per /home/pi/pi-timoloonfig.py

Please upgrade to ver 12.02.  This was a fatal bug that did not arise
during testing.  My Fault.  I have a note on the subDirChkMaxHrs about
doing some error trapping.  I will revisit.  In the meantime can you test
the change and let me know if it resolves your problem.  Basically the
function tries to find the most recent active subfolder using the datetime
name of the subfolder.  The check did not match the directory folder naming
minus the prefix.  YYYY-MMDD-HHMM.  I believe the error is fixed and you
should be OK

There is no Bug Bounty but I appreciate you notifying me.
Please confirm if the issue is resolved
Thanks Claude ...

pageauc, 2020-10-24:
Release 12.03 fixes trigger for creating new subdirectory when the max hours is exceeded.  Had to convert to float to get fraction of hours.  tested and works.   Please upgrade to new release using menubox.sh UPGRADE menu pick
Will leave this issue open.  Let me know results of your testing.
Claude ....

Distortionist, 2020-10-24:
Thank you, it seems to work now when I run the pi-timolo.py, I don't think it started them menubox.sh unless it quits it when i exit menubox.sh, I have not really spent time with it yet. I noticed I have really messed up the configuration. I did not realise the paths are relative to the folder set way later in the configuration. I know I screwed up when I reached it but I tried it anyway. It ended up making a folder named ~ in there because I used absolute paths for all the paths like ~pi/timelapse.

Silly me.

Ill reconfigure and I will let you know.

Thank you very much.

Distortionist, 2020-10-24:
This is missing, just gives a warning. I added this line to get rid of it. It is just missing from the settings for some reason.
PANO_DAYONLY_ON = 'False'    # Only take panoramic image during daytime

Other than than, a quick test everything works perfectly. I started and stopped everything from menubox, the webserver seems to work fine, when I go to recents then timelapse I get a page error but directly from timelapse it works. maybe it just needs time. I will update later if it doesn't work once I put it to work for a while.

Thanks again.

Distortionist, 2020-10-24:
Yes there is something not working with the recent folder in the webserver, if I browse to motion, it works, if i try to browse to timelapse, it gives an error.

pageauc, 2020-10-24:
Recent Folder can sometimes get screwed up if symlink points to a non existent file,  Recent entries are symbolic links to the actual images.  Once the recent max is reached oldest entry is removed and new symbolic link created for the latest entry. 

I suggest you delete all files in the recent folder (they are only sym links so the original image files will remain.  Example below assumes default location of recent folders.

eg
    cd ~/pi-timolo/media/recent/timelapse
    rm *  

Recent will then start to fill until max is reached.  Default is 40 but you can make this any reasonable value.  I don't recommend anything over 400 or 500 since it slows directory refresh.

PANO_DAYONLY_ON is just a new setting I was working on  but not implemented.  It is in the pi-timolo.py code but not in the config.py since it is not implemented yet.  Just ignore for now.

Messed up settings.  There is a config.py.prev so you could copy it over config.py

eg

    cp config.py config.py.bak
    cp config.py.prev config.py

Distortionist, 2020-10-25:
Ok thank you for your quick response.
I tested it out today and everything worked well. I left PANO_DAYONLY_ON in
there set to False, it does not seem to cause me issues so far.  I will
remove it if it does, I did notice some little things, I will verify
tomorrow if they happen again, but I noticed
makevideo.sh is not smart enough to figure out the files are in a folder
inside of timelapse, I manually moved them to timelapse, one folder below
where they were, and cleared the recent links manually. After that the
video is created. I am not sure if this is intentional and we need to edit
the video.conf with the folder containing the files we want to make a video
off each time. It is not a big deal at all, but it would be nice if it
figures out there is a folder in there and does it itself, and if there are
multiple, maybe list them and you select 1 2 3 or 4, whatever.

I have some requests other ideas, I am not sure how complex they are, and
I feel like I am asking a lot from you but they are just features that
would be extremely useful in your really cool software, you can think about
adding in your free time.

1) let us choose any destination for the video created, outside of the
media main folder, I would like it to go to a smb mount in mnt. It seems to
me tied to media folder.
2) let us choose custom exposure settings like ISO, Shutter, awb, etc... I
find the white balance all over the place it is what annoyed me the most,
but I would like control of all those if I can.
3) It would be nice if there were an option to lock ISO, shutter, awb, etc
based on the automatic settings that are computed at the first shot, and
maybe optionally to recalculate and lock every X shots.

I am not sure how hard all that is, but they would be extremely useful.

pageauc, 2020-10-25:
All Video settings are in the video.conf file.  You can change the folder
destinations Etc.for convid.sh and/or makevideo.sh

https://github.com/pageauc/pi-timolo/blob/master/source/video.conf

Disregard the avconf references since ffmpeg is auto detected. If you have
an old os distro then avconf will be the default
but later distros now use ffmpeg.  You should be able to do what you want
using the video.conf file.
Let me know if there are changes you would like.  I used settings to
automate putting timelapse videos onto a Kodi RPI so
I could easily review them on my TV.  This was a while back when we were in
Texas but Looks like we are stuck in Canada this winter.

I will take a look at manual override of camera settings.  It should not be
too difficult.  Not sure if there needs to be
individual variable settings (It might Clutter up config.py) ,  just a
parameter string, dictionary file or possibly a YAML config file.
Give me a list of the settings you would like to see overridden.

Distortionist, 2020-10-25:
I'll try the settings. I assumed it was like the pi-timolo.py where the
paths were locked in. I tried putting ~/something and it created a folder
called ~ in media haha. I'll set the path to the /mnt/ folder and see if it
works. The archive path worked.

Yes a string would be great, when set it can use them instead. One for
daytime and one for night time if possible.

I'm in Canada too, Hello from Nova Scotia.

pageauc, 2020-10-25:
We are in Ontario just a 10 min walk from Lake Huron.  Will see what I can
do about camera settings manual override.
Let me know how the video.conf works for you.
Claude ...

pageauc, 2020-10-25:
Ok After doing more testing It appears the subdirectory creation was not being triggered due to difference not being a float value.
I have done some quick testing and the subdirectory creation looks like it is triggered correctly when the age of the directory exceeds the specified max hours.  The check is used for timelapse and motion.

Please upgrade to release 12.03 using the menubox.sh UPGRADE menu pick
Please let me know the results of your testing
Than Claude ....

pageauc, 2020-10-25:
I released pi-timolo ver 12.04 that adds logic for PANO_DAYONLY_ON setting.  Also updated config.py but you can manually add variable if necessary otherwise message will appear.
Also
video.conf has settings for auto removing or moving source timelapse images to archive location 
There are also settings for copying video to a mount point share.  This is what I use to put my video's on a disk attached to a RPI running KODI.
I am looking at the manual override settings.  I think I will put settings in a python list (or maybe dictionary) in config.py.  One for Day and One for Night.  That way camera settings can be added or removed from the list as necessary.
Stay Safe and Healthy
Claude ...

Distortionist, 2020-10-25:
I have configured the settings appropriately. Thank you. I appreciate the
trouble adding the manual override as well.

I will upgrade later on today.

pageauc, 2020-10-25:
I have written a little python script that reads dictionary override values for the picamera.  It is not too long but demonstrates
applying custom settings.  The values on the sample dictionary settings are NOT very good but I put  variables down that I thought might be relevant.  Current settings do NOT take very good images but was just testing logic for reading camera vars.  
Can you work with this to get the type of images you would like.  I can use these as default override settings.

For picamera API Docs See https://picamera.readthedocs.io/en/release-1.13/api_camera.html#picamera

Code should be fairly easy to understand.  It is the camera settings that need tuning.

Thanks Claude ...

[testcam.zip](https://github.com/pageauc/pi-timolo/files/5435862/testcam.zip)

Distortionist, 2020-10-25:
I would love to try the Panoramas, can you recommend a motor and controller
for me to try that you know will work with your script?

I downloaded and checked out your override script. I think it has all the
settings I would like control of if I need. Thank you. I feel silly but
does pi-timolo automatically enable it if CAMERA_OVERRIDE_ON = True ?

It is terrible weather today so I really did not run the camera today.

pageauc, 2020-10-25:
This is currently a standalone script for testing purposes.  The idea would be to put the new variables with comments in config.py.
If you want to test pano I would recommend  waveshare hat since Pimoroni hat has to be ordered from England with currency exchange and extra shipping costs.  BTW I have several pimoroni hats and they are IMO better than Waveshare. 

Waveshare HAT
 https://www.amazon.ca/gp/product/B07PPV122Z/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1
Pimoroni HAT
https://shop.pimoroni.com/products/pan-tilt-hat?variant=22408353287

pi-timolo will automatically install the required pantilthat driver for either of these or a compatible.  

This hat sits on top of the RPI plugged into the GPIO header.  I also put standoffs on the opposite side to support the hat but not absolutely required.  See wiki for details https://github.com/pageauc/pi-timolo/wiki/Panoramic-Images-Stitching-Feature

Note assembly is a little finicky but there are YouTube videos (see wiki for a link) 

BTW I would still like to see what settings you come up with for the camera override settings.
Claude ....

Distortionist, 2020-10-25:
My HQ camera is new I only did that one timelapse test with it so far, and
just mucking around inside learning how to use it and the lenses. I
currently have 2 pi cameras 2.1 stuck to a window with tape taking 24 hour
timelapses with simple bash scripts I made that use the sunrise and sunset
as timers with an offset of 35minutes.

In daytime I was not happy locking exposure settings, the dark and shade vs
bright sun was too much to find a happy median that didn't cause seizures.
It was too annoying so it currently is not locked to any exposure, but I am
trying locking to different awb settings. I think I like cloud setting the
most.

The night time camera is a NoIR camera to catch the stars moving in the
night sky (or see the result of earth spinning I guess), I have the awb
locked to greyworld, the ISO locked to 800, and the shutter locked to 10
seconds which is the maximum.

I expect I would do the exact same with the HQ camera except the night
shutter speed I would probably go to 15 seconds, I would also experiment
with 20 seconds but I don't think I would like the trails from the stars.

I will test that script when I get the chance.

pageauc, 2020-10-28:
FYI.  Release 12.06 fixes a problem not maintaining TIMELAPSE_RECENT_MAX.  It now keeps timelapse recent at specified maximum.  I think it occured when global variables were renamed to capitalized snake format.  Works now and recent is much faster with smaller number of recent entries.  I set mine to 100 but default is 40.  Anyway it now works the way it should. It should autocleanup older recent entries.  

Hope everything is going well and you are safe, healthy and happy
I will add the image settings override feature when I get your recommended default settings.
Added a few more settings.  See zip file
 
[testcam.zip](https://github.com/pageauc/pi-timolo/files/5451558/testcam.zip)

Claude ...

Distortionist, 2020-10-28:
Hi there! All is well thank you. I hope things are going well for you too.
Sorry I have not gotten back to you, I gave the script a good run the day
before yesterday for the first time without the overrides to have a base to
compare to. Unfortunately it has been pretty crappy and rainy outside,
today started with a bit of snow but it was very little and it may be
clearing enough to give the overrides a try outside. The white balance and
timelapse was better  this time but it did start jumping around a bit as it
got darker before it settled on good night settings. I have linked my
videos below with 2.1 cameras I am using currently with my own simple
raspistill scripts, the nighttime ones are locked everything to manual
settings, the day one is only locked to awb cloud, I may lock the ISO to
100. The night settings I have currently locked are ISO to 800, Shutter to
10 seconds, and awb to greyworld because it is a noir cam.

I will update pi-timolo before running it again and I will check out the
new overrides while I am at it.  I got a chance to upload all the videos I
have made with timelapses. My other pi with 2 cameras stuck to my window
taking 24h timelapses, one camera is for day and one is for night. It is
driving me crazy though it won't switch automatically in a script, it just
hangs but if i invoke the script rather than let it run from crontab it
runs fine. Oh the joys. Anyway, that is another project I am working on at
the same time.

I ordered the Pimoroni hat. Not sure when I will get the chance to play
with it, I would like to mount the HQ camera to it I am using with your
pi-timolo. I would have to figure out how to mount it on my tripod. I have
a pretty crude setup. I may have to buy a different camera for it. I am not
sure if it can take the HQ cameras weight.

I managed to upload all my videos to youtube, this is all new to me,
including my weather station and website you can see here
http://livingstonecove.ca
The videos I uploaded including the pi-timolo test can be found here
https://www.youtube.com/channel/UCKGTTzMHcEtq-U22-Twfv9A
The pi-timolo test is https://www.youtube.com/watch?v=_bTeVI7dUWE

I will let you know when I do the override test.

I did find that the video maker cleanup misses one file haha. It is pretty
funny but I have the path for the source set to media/timelapse, so I move
the images from the dated subfolder created making the timelapse with mv *
.. and make the timelapse. After it is done it cleans up and moves
everything to archive except for 1 file. I manually delete it. I also
manually clear the recents broken links after that.  I feel like this is
not the way you intended me to use it. I think you intended users to change
the video.conf each time with the foldername of the timelapse shots I want
to turn to a video. It is just easier and quicker to move and not edit the
file. Maybe if it automatically figured out the folders and went in and
made a video for each folder it would be more convenient, or maybe I am
just missing something here.

Cheers and thank you.

pageauc, 2020-10-28:
You should add music to your videos.  You can do this from within YouTube Studio on a PC with chrome browser.
See https://support.google.com/youtube/answer/94316?hl=en

This is a mini video editor.  You can add an audio track and pick creative commons music or sound effects from the google audio library.  They have a large library of pretty good music.

I normally edit in my videos using my PC video editor. A good free Open Source video editor is Lightworks https://www.lwks.com/
Claude ...

Distortionist, 2020-10-28:
I usually edit my videos from my PC and add sound but I don't really do
much editing. These timelapses I am trying to make completely automated, I
am self employed and a single dad with too many things on the go to edit
these videos and pretty them up manually. I am always playing catch up with
everything I have to do! I just love to see the day go by fast, my
favourite are the night ones, I am patiently waiting for a nice long
cloudless night to see the stars go by. I love that NOIR camera and the
night time lapse videos it produces. I also can't wait to use my HQ camera
I use with pi-timolo on a nice clear night where I can leave it out there
pointing up and taking 15 second exposure pictures all night. I am
interested to see its bigger sensor and lens pictures compared to the
smaller 8mp camera with no IR filter. I really want to catch meteors and
satellites in the sky. I love that stuff.

Distortionist, 2020-10-29:
I feel really dumb but I am trying to test the override script, and it
looks like it takes a shot, but I have no idea where on earth it put it!
It is not in the folder I ran it from, it is not in the media/timelapse
folder, not in the pi-timolo folder.. haha. Sorry but can you point me in
the right direction here?

pageauc, 2020-10-29:
I sent you a copy of testcam.py that I was testing with.  I set
    CAMERA_OVERRIDE_DAY_ON = False   # Default= False Override Day Settings Only (daymode auto detects Day)
    CAMERA_OVERRIDE_NIGHT_ON = False

This turned off the override feature so you did not get any override.  I have added a reminder at bottom of script.  Here are the settings currently

```
CAMERA_OVERRIDE_ON = True     # Default= False Do NOT use Override settings True= Enabled
CAMERA_OVERRIDE_DAY_ON = True   # Default= False Override Day Settings Only (daymode auto detects Day)
CAMERA_OVERRIDE_NIGHT_ON = True # Default= False Override Night Settings Only (daymode auto detects Night)
CAMERA_WARM_SEC = 4           # Default= 4 Seconds to wait for camera to warm up

daymode = True  # This var will normally be set in pi-timolo based on pixel average.
                 # change to appropriate value for testing.
```

Please use the attached script that should work better for you.   It just does one image at a time by design.  I did not want to put in a loop since it would required adding more code.
Claude ....

[testcam.zip](https://github.com/pageauc/pi-timolo/files/5459351/testcam.zip)

pageauc, 2020-10-29:
I think I will eliminate the CAMERA_OVERRIDE_ON setting since they day night overrides would be simpler and less confusing.
Note this script is just for testing things.  Also some camera values can be read from the actual value the camera used.  I may display the actual camera setting used where it is practical.  Eg if Automatic setting used it would show what the camera AWB, Shutter, Etc was actually used by the camera.  This is covered in the picamera api docs. 

See https://picamera.readthedocs.io/en/release-1.13/api_camera.html#picamera

Claude ...

Distortionist, 2020-10-29:
Yes it produced a picture now.

I did change the settings but I only changed CAMERA_OVERRIDE_ON to True. I
did not think I needed the other two because it was just one picture it
produces at a time. I turned them  all on and it works as expected now.
With our luck the clouds took over already and I don't know If it will look
all that great, but I have some settings to try. I will try them anyway for
a little bit and let you know!

Distortionist, 2020-10-29:
Here is the short test today, it does look better than the last one but it
looks like it is going to rain and I got the camera back inside. I need to
test it over a full day to see how the whitebalance and shadows behave.

Here are the options I used for todays test :

CAMERA_DAY_OVERRIDE = {'resolution':(1920, 1080), # W, H of image to capture
                       'hflip':False,     # True or False
                       'vflip':False,      # True or False
                       'rotation':0,      # Valid values in degrees  0, 90,
180
                       'iso':100,         # Valid values 100 to 800
                       'brightness':50,   # Valid Values 1 to 100
                       'contrast':20,      # Valid Values -100 to +100
                       'saturation': 5,    # Valid Values -100 to +100
                       'sharpness':30,     # Valid Values -100 to +100
                       'shutter_speed':0, # 0=Auto or microseconds value
                       'awb_mode':'cloudy',
                       'exposure_mode':'auto',
                       'image_effect':'none',
                       'meter_mode':'matrix'
                      }

And this is the result :

https://youtu.be/H9tqvcckjzc

The weather conditions are quite different than last time but it does seem
less jumpy with the clouds, if you compare to last video around the 1
minute 10 second mark. I will test again when I get the chance for a whole
day. I should have used shorter intervals as well, I forgot to change the
warmup timer.

Distortionist, 2020-11-08:
Hi Claude. I hope you are well. I have received the pimoroni pan tilt hat.
I can't play with it because the only free camera is the HQ camera and it
doesn't fit the mount. I have to investigate finding a new mount although
it may just not be able to take the HQ camera's weight. The 2 cameras I
have that would fit are stuck to my window taking time lapses already.

I have not had much chance to play with the test, the wind has been strong
and we have had cloudy weather. Today is nice and it is currently shooting.
I did however fine tune the other cameras, and to fix my white balance
issues I have switched to manual control. There are 2 settings that would
be useful to take control off and I really find they make a huge difference
are the manual white balance, in raspistill it is a -awbg switch and then
you define red and blue gain (green is fixed). For daytime on the 2.1
cameras locking them at 2.0,1.3 produces nice color throughout the day so
far, and the white balance is not jumping around. I also started using
dynamic range compensation, so when its a bright sky it doesn't turn the
landscape black, it would try to dim the brighter areas and brighten up the
dark areas, it works really well. My first test was yesterday, the results
are awesome. I had turned the saturation a bit higher with the new settings
to see if I get better results now but I had to tone it down for today, but
I am very happy with the results.

If you can add those to the test it would be awesome, I find they improve
the time lapse quality significantly! Here is a still from today with the
locked white balance and the dynamic range compensation. It is a MASSIVE
improvement in a still thanks to the DRC and the video is improved
significantly with consistent white balance.
[image: image.png]

closed 2021-04-24 by pageauc