pageauc/pi-timolo #110: Python3 compatibility

surak, 2020-12-19:
While pi-timolo is trivial to work with python3, the webserver.py is not. 

Since python2 is deprecated and unmaintained, I would humbly suggest we ditch it and go with python3 from here on.

pageauc, 2020-12-19:
I try to make my code python 2 and 3 compatible.  I will take a look at the
webserver and make it python2 and 3 compatible.
Claude ...

surak, 2020-12-19:
Thanks, Claude. I made a merge request (#111) which will work only in Python3 (no module html in python2), but one can easily do a try import html, and if it throws an exception, load cg as html, for example.

pageauc, 2020-12-20:
Thanks, I think python3 only is fine since mostly standard libraries.  I
will keep pi-timolo.py python2 and 3 compliant because it uses OpenCV
although buster allows importing python3 opencv using apt-get.
If this is stable can you change the PROG_VER to 12.0 and I will do a
merge.  I appreciate your contribution.  I was going to write update but
would have done it using try except to make it dual compatible but like I
said I don't think python 3 only is an issue.
Please send me an updated push with ver change and any other updates you
may have.  You can add your credit in the code comments.
Thanks Claude.

pageauc, 2021-01-11:
I ran across issues with Jessie so made python2 ver the default.  Added instructions to copy webserver3.py to webserver.py.  This will allow webserver.sh and menubox.sh to work correctly for python3.
Your input is appreciated.

I have had experience with trying to make a robotic servo driver python2 and 3 compatible.  A lot of try except's and very messy code.  Thought it was cleaner to just have a clean webserver3.py that can be implemented on Buster.  At some point I will reverse the instructions to make python3 version the default and instructions to fall back to python2 webserver.   I still have a some RPI's running Jessie so there are users still running older versions of Debian.
Let me know if this is a suitable implementation plan.
Claude ...

closed 2021-01-11 by pageauc