pageauc/pi-timolo #116: Daemon is crashing

tibaultbordeaux, 2021-03-12:
Dear Claude,

I tried pi-timolo again after a long time, but without success.
I'm on a Raspberry Zero W with a fresh RaspberryPi OS Buster Full
sudo apt-get update/upgrade was done
then installed latest version of pi-timolo

pi-timolo.py and webserver.py are working fine.
But when using *.sh to start the daemons, the background tasks are crashing immediately:

```
pi@rpi-zero:~/pi-timolo$ ./pi-timolo.sh start
./pi-timolo.sh ver 1.1 written by Claude Pageau
-----------------------------------------------
START   - Start /home/pi/pi-timolo/pi-timolo.py in Background
STATUS  - Waiting 10 seconds ....
STATUS  - pi-timolo.py is Running ...
STATUS  - PID is 2345
INFO    - To Stop pi-timolo.py execute command below
INFO    - ./pi-timolo.sh stop
pi@rpi-zero:~/pi-timolo$ ./pi-timolo.sh status
./pi-timolo.sh ver 1.1 written by Claude Pageau
-----------------------------------------------
STATUS  - pi-timolo.py is Not Running ...
INFO    - To Start pi-timolo.py execute command below
INFO    - ./pi-timolo.sh start
```

Do you have any hints for me?

pageauc, 2021-03-12:
run *pi-timolo.py* directly and see what log messages appear.  This will
allow you to diagnose the problem.
May be due to a library issue or camera problem

*cd ~/pi-timolo*

*./pi-timolo.py*

If you cannot diagnose problem from log messages then send me log and I
will help

Claude ...

tibaultbordeaux, 2021-03-13:
Dear Claude,
thanks for your fast reply!
I did try running pi-timolo.py before and it works flawlessly.
I do not get any error messages. Same with webserver.py.

Probably it wasn't a good idea to update my original post, as you might not have seen the update:

Adding "python" in front of the command in pi-timolo.sh like this ...
```
python $progpath/$progname >/dev/null 2>&1 &
```
is a working solution for me. However, this does not seem to work within webserver.sh. It's wierd.

tibaultbordeaux, 2021-03-13:
Ok, maybe I'm getting closer.
When starting ./pi-timolo.py within the directory pi-timolo erverything is worling fine.

But when starting from outside of this directory, i.e. by using
/home/pi/pi-timolo/pi-timolo.py

I get this error:
```
[Errno 2] No such file or directory: 'media/shortl'
```

tibaultbordeaux, 2021-03-13:
Ok, here is my next update.
I got rid of the above error message when using the full path in config-file, i.e.
/home/pi/pi-timolo/media/shortl

Next step was adding "python3" instead of "python" in webserver.sh.

The situation now is like this:
I still get 
```
STATUS  - pi-timolo.py is Not Running ...
```
but: pi-timolo is working now.

pageauc, 2021-03-13:
If you change the location of the pi-timolo install directory you will need
to change some bash script variables
eg  in pi-timolo.sh.  In order for bash scripts to run pi-timolo.py in the
background it needs an absolute path
since background shell may not have a path variable.  That means some of
the shell scripts would
need the progpath variable edited.  This is the default for
*pi-timilo.sh* where
*progpath* variable
would need to be edited.  Note the quotes must be included and no space
after the =

progpath="/home/pi/pi-timolo"
> progname="pi-timolo.py"

Here is the default for *webserver.sh* where *progpath* variable would need
to be edited

progpath="/home/pi/pi-timolo"
> progname="webserver.py".

Note these scripts are only used to start background tasks.

For more info on absolute vs relative paths see
*https://www.computerhope.com/issues/ch001708.htm
<https://www.computerhope.com/issues/ch001708.htm>*

If you think things are really messed up, you may want to consider deleting
your current installation and do a fresh install.
and run from default location.  If you want to run pi-timolo from a mounted
hard disk or thumb drive see wiki
*https://github.com/pageauc/pi-timolo/wiki/How-to-Run-Pi-Timolo-from-Another-Directory-Path
<https://github.com/pageauc/pi-timolo/wiki/How-to-Run-Pi-Timolo-from-Another-Directory-Path>*
Let me know how you are doing.
Stay Safe
Claude ...

closed 2021-04-24 by pageauc