pageauc/pi-timolo #117: Ideal settings for Raspberry Pi Camera V2 (8 Mega Pixels) for motion detection

npaibrooklyn, 2021-04-22:
Hello @pageauc 

I have a Raspberry Pi Camera V2 (8 Mega Pixels). https://www.raspberrypi.org/documentation/hardware/camera/

What settings do you recommend for these?
IMAGE_WIDTH = 1280           # Default= 1024 Full Size Image Width in px
IMAGE_HEIGHT = 720           # Default= 768  Full Size Image Height in px

Can I use the max?  3280 and 2464 pixels for width and height respectively?

I will be using this for motion detection (mostly in day time). Any other settings you recommend for this camera for this purpose (especially the MOTION_TRACK_ *settings)?

Thanks in advance - I love your project!

- Nandan

pageauc, 2021-04-22:
Nandan

Yes you can use the maximum.  pi-timolo detects pi-camera module version
and corrects any resolution issues.  eg exceeding max resolution of the
specific camera moldule version.  This can be found on line 482 of
pi-timolo.py.
Only issue is having to scroll up/down on webserver web page image to see
other parts of image.  This also happens with eg 1080p size image setting.
I might look at adding feature to auto resize webserver  display image.

Stay Safe
Claude ....

npaibrooklyn, 2021-04-22:
Understood, thank you for the quick answer @pageauc . Another follow up question, if you don't mind:

Given your experience and for my camera, what values do you recommend for these settings if I am going to be using the camera at a bird-feeder to motion-detect birds (mostly during the day). Should I leave them as is or are there some other baseline settings I could begin experimenting with?

STREAM_WIDTH = 320           # Default= 320  Width of motion tracking stream detection area
STREAM_HEIGHT = 240          # Default= 240  Height of motion tracking stream detection area
STREAM_FPS = 20              # Default= 20 fps PiVideoStream setting.  Single core RPI suggest 15 fps
STREAM_STOP_SEC = 0.7        # Default= 0.7 Allow time to stop video stream thread to release camera

# Motion Track Settings
# ---------------------
MOTION_TRACK_ON = True         # Default= True True=Turns Motion Detect On, False=Off
MOTION_TRACK_INFO_ON = True    # Default= False Hide detailed track progress logging messages
MOTION_TRACK_TIMEOUT_SEC = 0.3 # Default= 0.3 seconds Resets Track if no movement tracked
MOTION_TRACK_TRIG_LEN = 75     # Default= 75 px Length of motion track to Trigger motionFound
MOTION_TRACK_MIN_AREA = 100    # Default= 100 sq px  Minimum Area required to start tracking

pageauc, 2021-04-22:
Standard motion track settings should work OK depending on how close birds
are to camera.  My feeder was in front of the house where it was getting
triggered by cars going by so lowered feeder/ raised camera to background
was not pointing at road and picking up sidewalk/road objects.  If you are
doing stills I recommend enabling

MOTION_TRACK_MINI_TL_ON = False     # Default= False  True= Take a mini
> time lapse sequence rather than a single image (overrides MOTION_VIDEO_ON)
> MOTION_TRACK_MINI_TL_SEQ_SEC = 30   # Default= 30 secs Duration of
> complete mini timelapse sequence after initial motion detected
> MOTION_TRACK_MINI_TL_TIMER_SEC = 5  # Default= 5 secs between each image.
> 0 is as fast as possible
>

Adjust the seq_sec and timer_sec to suit.   This will take mage every
specified seconds for the sequence sec duration.  That way you get birds
after they land.  30 seconds may be too long so you might want to set for
shorter time.  Just adjust until you get good sequences after motion
tracking is triggered.  This should avoid getting only one blurry image of
a bird in flight.  The sequence should result in better selection of images.

Let me know how you make out.

Stay safe.  Claude ....

pageauc, 2021-04-22:
Note you can also try motion tracking video mode settings as well.  Note
setting on previous email should set

MOTION_TRACK_MINI_TL_ON = True.
>

npaibrooklyn, 2021-04-22:
That's a great suggestion @pageauc . I added code to userMotionCode(filename) to make some some calls to cloud based APIs. When a picture is taken, before the next one can be taken, does userMotionCode have to completely finish? Or is that run in a thread? I am worried that the API calls may take some time and a quick (low value of MOTION_TRACK_MINI_TL_TIMER_SEC ) sequence of time lapsed pictures may be hindered?

pageauc, 2021-04-22:
userMotionCode takes place after full sequence is done.  If necessary you
could start your cloud based api calls in a thread as part of the
userMostionCode.
It would be best if you put the api calls in an executable bash or python
script and just call in a thread from userMotionCode.  The last image will
be what is passed to the userMotionCode.  Your api script should be setup
to read the image filepath so it can send to cloud api.

Hope this helps.

npaibrooklyn, 2021-04-22:
I am very grateful for all your suggestions. I will try all of them out and share my findings! Great project once again!

pageauc, 2021-04-22:
There are other ways to do this but here is a simple way to spawn a
subrocess.  This is an example
user_motion_code.py file.  Put your cloud api code in a file called
cloudrun.py in the pi-timolo folder per code run_cmd variable in code below
You can modify this to suit.

# run cloudrun.py in a separate process after pi-timolo usermotion
> processes are complete
> # The filenamePath will be the last image or video file saved by pi-timolo.
>
>
> #------------------------------------------------------------------------------
> def userMotionCode(filenamePath):
>     run_cmd = '/home/pi/pi-timolo/cloudrun.py' + ' ' + filenamePath
>     try:
>         logging.info("Run userMotionPath Command per Below")
>         print("%s" % run_cmd)
>         # spawn stitch command with parameters as seperate task
>         proc = subprocess.Popen(run_cmd, shell=True, stdin=None,
>                                 stdout=None, stderr=None, close_fds=True)
>     except IOError:
>         logging.error("Failed subprocess %s", run_cmd)
>

Here is example of running a python script
https://www.tutorialspoint.com/python/python_command_line_arguments.htm

You will only need to read one parameter in  cloudrun.py so your code can
be simpler see untested cloudrun.py
code below as a sample or reading the filenamePath from code above.

#!/usr/bin/env python
>
> import sys
>
> if len(sys.argv) != 2:
>     raise ValueError('filenamePath must be passed as a parameter from
> user_motion_code.py')
>
> filenamePath = sys.argv[1]
>
> # Add your cloud api stuff here and use filenamePath variable to process
> the last motion image passed by user_motion_code.py
>

Make sure cloudrun.py file is executable using the
*chmod **+x cloudrun.py*

This is untested sample code but should give you enough to get started

Stay Safe.  Claude ....

pageauc, 2021-04-24:
Please note pi-timolo.py issues with not properly recognizing V2 picamera and image quality in writeTextToImage function have been fixed in Release 12.11.  You can upgrade using menubox.sh UPGRADE option.  I will close this issue but let me know how you make out with your cloud API motion detection.  Regards Claude ....

closed 2021-04-24 by pageauc