pageauc/pi-timolo #118: Camera version is being detected as 1.3 even though 2.1 version is connected

npaibrooklyn, 2021-04-22:
Hello @pageauc I just noticed that pi-timolo is detecting the camera version as 1.3 even though I have connected a 2.1 version of camera.
This is what I am seeing on the console:
2021-04-22 16:23:09 INFO     <module>   Pi Camera Module is Enabled and Connected supported=1 detected=1
2021-04-22 16:23:09 INFO     <module>   picamera version is 1.13

As a result, I am not able to use the max resolution (3280 × 2464 pixels) and it is rounding it off to (2592 × 1944 pixels)

Looks like the version is detected thusly:
picameraVer = require('picamera')[0].version

Is is merely looking at the module's version?

npaibrooklyn, 2021-04-22:
Some more info: 

$ pip freeze | grep picamera
picamera==1.13

Perhaps I need to just upgrade the module?

npaibrooklyn, 2021-04-22:
@pageauc doesn't look like there is a 2.1 version of the picamera module. Perhaps we need a different way of detecting the camera version?

npaibrooklyn, 2021-04-22:
@pageauc I think I have a possible fix. I tested it and it works. Please feel free to take this for your repo and alter it as you wish, if you like the fix:

https://github.com/npaibrooklyn/bird-brain/pull/1/commits/f9a3cb8ddd01f5e014c4f5b1f41b2f03306497b0

pageauc, 2021-04-23:
After some work I found a cludge for checking the maximum resolution of the
attached picamera module hardware

This is the code i wrote based on some forum threads

import os
>
> print('Checking Pi Camera Module version Wait ...')
> os.system('/usr/bin/raspistill -o ./image.jpg')
>
> from PIL import Image
> from PIL.ExifTags import TAGS
> img = Image.open('./image.jpg')
> exif_data = img._getexif()
> for tag, value in exif_data.items():
>   if TAGS.get(tag, tag) == 'ImageWidth':
>       imageWidth = value
>       if value >= 3280:
>           print('Max ImageWidth is %i PiCamera module hardware is ver 2' %
> value)

> #  print TAGS.get(tag, tag), value
> os.remove('./image.jpg')
>

This works under python2 and 3 and uses raspistill to take an image, then
exif data fow imageWidth is used
to determine the max resolution width of the attached picamera module.
Note this does not check for a ver1 camera module. but could just add else

Would like a more elegant way to check the camera hardware version.  Looks
like the require method fails.

I will look at updating pi-timolo to use the above method to check for the
camera module hardware version

Claude ...

pageauc, 2021-04-23:
I have updated pi-timolo.py to ver 12.07 that now uses raspistill and exif
data to determine what hardware version of pi camera is installed
Please run menubox.sh UPGRADE option and test.  Let me know results
Thanks Claude ....

npaibrooklyn, 2021-04-23:
Hey @pageauc instead of creating an image and checking its property (which as you pointed out is not elegant) you can use the method that pycamera python module supplies. I have shown it in https://github.com/npaibrooklyn/bird-brain/commit/f9a3cb8ddd01f5e014c4f5b1f41b2f03306497b0

npaibrooklyn, 2021-04-23:
with picamera.PiCamera() as cam:
        CAM_MAX_RESOLUTION = cam.MAX_RESOLUTION
        logging.info("Cam max resolution is %s", CAM_MAX_RESOLUTION)
        CAM_MAX_WIDTH, CAM_MAX_HEIGHT = CAM_MAX_RESOLUTION.width, CAM_MAX_RESOLUTION.height
        logging.info("Cam max width is %s, Cam max height is %s", CAM_MAX_WIDTH ,CAM_MAX_HEIGHT)

is the code block to get the max width and max height directly from the camera. Pls refer to https://github.com/npaibrooklyn/bird-brain/commit/f9a3cb8ddd01f5e014c4f5b1f41b2f03306497b0

npaibrooklyn, 2021-04-23:
@pageauc if you wish, you can give me collaborator access to your repo and I can submit the fix above in a PR for your review. I believe it is more elegant and faster than writing and image and checking its property. Or you can just copy the changes from  https://github.com/npaibrooklyn/bird-brain/commit/f9a3cb8ddd01f5e014c4f5b1f41b2f03306497b0

npaibrooklyn, 2021-04-23:
Thanks for making the fix @pageauc . Just one more thing: we can get rid of setting picameraVer variable to 1 or 2 and remove its usages. We have the resolution of the camera so we can just that directly. You can see the changes in https://github.com/npaibrooklyn/bird-brain/commit/f9a3cb8ddd01f5e014c4f5b1f41b2f03306497b0

The relevant block is pasted here as well.

```
# Round image resolution to avoid picamera errors
image_width = (IMAGE_WIDTH + 31) // 32 * 32
if image_width > CAM_MAX_WIDTH:
    image_width = CAM_MAX_WIDTH
logging.info("Using image width %s", image_width)

image_height = (IMAGE_HEIGHT + 15) // 16 * 16
if image_height > CAM_MAX_HEIGHT:
    image_height = CAM_MAX_HEIGHT
logging.info("Using image height %s", image_height)

stream_width = (STREAM_WIDTH + 31) // 32 * 32
if stream_width > CAM_MAX_WIDTH:
    stream_width = CAM_MAX_WIDTH
logging.info("Using stream width %s", stream_width)

stream_height = (STREAM_HEIGHT + 15) // 16 * 16
if stream_height > CAM_MAX_HEIGHT:
    stream_height = CAM_MAX_HEIGHT
logging.info("Using stream height %s", stream_height)

```

closed 2021-04-24 by pageauc