pageauc/pi-timolo #119: Motion tracked images stored in media folder are not high definition

npaibrooklyn, 2021-04-23:
Hello @pageauc Completely unrelated to other issues I reported, I noticed that motion-tracked (mo-*) images in the media folder (both media/motion and media/recent) are very small (600KB for example). I was running pi-timolo (12.06 version) without any changes to config.py. I used ./pi-timolo.py to run it. My python version is 2.7.16. My camera is a Raspberry camera v2 (**8 Megapixels**). Shouldn't the image be a better quality?

pageauc, 2021-04-23:
As part of postImageProcessing the saved image is reopened to add text and
re-add exif data.  I believe PIL is doing the image compression.  I will
add quality=
to the img.save if in jpg format.  PIL was saving with default quality and
that is why size was smaller.  Setting quality to 95 increases jpg file
size.
If you save as png format file sizes are a LOT larger but honestly I cannot
tell the difference in quality. between compressed jpg and png.

Also tried V2 camera at highest resolution.  First TL image was OK but got
error on second.  gpu_mem is 128.  Looks like some gpu memory issue
after first successful image.  Not sure why.  Needs more testing

mmal: mmal_vc_component_enable: failed to enable component: ENOSPC

npaibrooklyn, 2021-04-23:
I tried with jpeg quality to 100 with all the other config settings set to default, I still got smaller size images. I can turn off postImageProcessing and test.

pageauc, 2021-04-23:
.
Update github pi-timolo.py to 12.10 version
This update implements your method of checking max camera resolution.
Thanks. As I said the other method was a Cludge.
Tested on RPI4 Buster with V2 camera module,  RPI3  Buster with V1 camera
module and rpi3 Jessie with V1 camera module. Tested with both python 2 and
3.  No issues found

Also added jpg quality to PIL image save in writeTextToImage function.
This does increase image sizes and quality appears better.
Also tested .png and they are a lot larger size but cannot see significant
quality difference for the size increase.

Can you please test this version and let me know.  You can upgrade using
menubox.sh UPGRADE option.

Your feedback is appreciated
Thanks Claude ....

npaibrooklyn, 2021-04-23:
Hello @pageauc I tested the postImageProcessing and it still creates small images (300KB in my test). When I commented out the call to writeTextToImage(..) , it was able to save the images at 4 MB.

npaibrooklyn, 2021-04-23:
Hello @pageauc I tested with 
img.save(imagename, quality="maximum") 
and it worked

Pls consider providing these presets in your config:
https://github.com/python-pillow/Pillow/blob/master/src/PIL/JpegPresets.py#L71

The presets provide control on subsampling and quantization and come with variety of options:
``web_low``, ``web_medium``, ``web_high``, ``web_very_high``, ``web_maximum``,
``low``, ``medium``, ``high``, ``maximum``

npaibrooklyn, 2021-04-23:
@pageauc I took further look into the PIL library code and this would be a better code:

```
    if IMAGE_FORMAT == '.jpg':
        img.save(imagename, quality="keep")
    else:
        img.save(imagename)
```

Note that quality of jpegs are guided by the compression (1-100), subsampling and quantization tables

Using quality="keep" ensures that all three are maintained as is from the original file. Refer to the PIL code here: https://github.com/python-pillow/Pillow/blob/master/src/PIL/JpegImagePlugin.py#L645
```
    if quality == "keep":
        quality = -1
        subsampling = "keep"
        qtables = "keep"
```

I thoroughly tested this and the image before and after postImageProcessing is the same except an additional few bytes due to the text.

pageauc, 2021-04-24:
Released ver 12.11 of pi-timolo.py with quality="keep" Change.  I agree this is a better option.  
Your suggestions are appreciated.
Claude ...

closed 2021-04-24 by pageauc