pageauc/pi-timolo #120: Permission Denied on Startup

pjockey, 2021-04-25:
This was working, and then all of a sudden, I'm getting this error:

```
pi@raspberrypi:~/pi-timolo $ python3 pi-timolo.py
-------------------------------------------------------
pi-timolo.py ver 12.06  written by Claude Pageau
-------------------------------------------------------
Loading Wait ....
2021-04-25 11:43:45 INFO     <module>   Pi Camera Module is Enabled and Connected supported=1 detected=1
2021-04-25 11:43:45 INFO     <module>   picamera version is 1.13
2021-04-25 11:43:45 INFO     <module>   No Plugin Enabled per PLUGIN_ON=False
2021-04-25 11:43:45 INFO     <module>   picamera ver 1.13 Max Resolution is 2592 x 1944
2021-04-25 11:43:45 INFO     <module>   Testing if Pi Camera is in Use
2021-04-25 11:43:48 INFO     <module>   Pi Camera is Available.
2021-04-25 11:43:48 INFO     <module>   Start pi-timolo per /home/pi/pi-timolo/config.py Settings
2021-04-25 11:43:48 INFO     timolo     Start PiVideoStream ....
----------------------------------- Settings -----------------------------------
Config File .. CONFIG_FILENAME=config.py  CONFIG_TITLE=pi-timolo ver 12.04 Default Settings
     Plugin .. PLUGIN_ON=False

Image Info ... Size=1280x720  ext=.png  Prefix=cam1-  VFlip=True  HFlip=True  Rotation=0
               IMAGE_GRAYSCALE=False   Preview=False
   Low Light.. NIGHT_TWILIGHT_MODE_ON=True  NIGHT_TWILIGHT_THRESHOLD=90  NIGHT_DARK_THRESHOLD=50  NIGHT_BLACK_THRESHOLD=4
               NIGHT_MAX_SHUT_SEC=5.90  NIGHT_MAX_ISO=800  NIGHT_DARK_ADJUST=4.70  NIGHT_SLEEP_SEC=20
   No Shots .. IMAGE_NO_NIGHT_SHOTS=False   IMAGE_NO_DAY_SHOTS=False
   Img Text .. On=True  Bottom=True (False=Top)  WhiteText=True (False=Black)
               SHOW_TEXT_WHITE_NIGHT=True  SHOW_TEXT_FONT_SIZE=18 px height

Motion Track.. On=True  Prefix=mo-  MinArea=100 sqpx  TrigLen=75-120 px  TimeOut=0 sec
               MOTION_TRACK_INFO_ON=True   MOTION_DOTS_ON=False  IMAGE_SHOW_STREAM=False
   Stream .... size=320x240  framerate=20 fps  STREAM_STOP_SEC=0.70  QuickPic=False
   Img Path .. MOTION_PATH=/home/pi/pi-timolo/media/motion  MOTION_CAM_SLEEP=0.70 sec
   Sched ..... MOTION_START_AT  blank=Off or Set Valid Date and/or Time to Start Sequence
   Force ..... MOTION_FORCE_SEC=60 min (If No Motion)
   Lockfile .. On=False  Path=/home/pi/pi-timolo/pi-timolo.sync  NOTE: For Motion Images Only.
   Num Seq ... MOTION_NUM_ON=True  numRecycle=True  numStart=1000   numMax=2000  current=1197
   Num Path .. NUM_PATH_MOTION=./data/mo-pi-timolo.dat 
   Quick TL .. MOTION_TRACK_MINI_TL_ON=False  Quick Time Lapse Disabled
   Video ..... MOTION_VIDEO_ON=False  Motion Video is Disabled
   Sub-Dir ... MOTION_SUBDIR_MAX_HOURS=0 (0-off)  MOTION_SUBDIR_MAX_FILES=0 (0=off)
   Recent .... MOTION_RECENT_MAX=40 (0=off)  MOTION_RECENT_DIR=media/recent/motion

Time Lapse ... On=True  Prefix=tl-   Timer=120 sec   TIMELAPSE_EXIT_SEC=0 (0=Continuous)
               TIMELAPSE_MAX_FILES=0
   Img Path .. TIMELAPSE_PATH=/home/pi/pi-timolo/media/timelapse  TIMELAPSE_CAM_SLEEP_SEC=4.00 sec
   Sched ..... TIMELAPSE_START_AT  blank=Off or Set Valid Date and/or Time to Start Sequence
   Num Seq ... On=True  numRecycle=True  numStart=1000   numMax=2000  current=1223
   Num Path .. numPath=./data/tl-pi-timolo.dat
   Sub-Dir ... TIMELAPSE_SUBDIR_MAX_HOURS=0 (0=off)  TIMELAPSE_SUBDIR_MAX_FILES=0 (0=off)
   Recent .... TIMELAPSE_RECENT_MAX=40 (0=off)  TIMELAPSE_RECENT_DIR=media/recent/timelapse

Disk Space  .. SPACE_TIMER_HOURS=0 (Disabled) - Manage Target Free Disk Space. Delete Oldest jpg Files
            .. Check Every SPACE_TIMER_HOURS=0 (0=Off)  Target SPACE_TARGET_MB=500 (min=100 MB)

Logging ...... VERBOSE_ON=True (True=Enabled False=Disabled)
   Log Path .. LOG_TO_FILE_ON=False  LOG_FILE_PATH=/home/pi/pi-timolo/pi-timolo.log
--------------------------------- Log Activity ---------------------------------
2021-04-25 11:43:51 INFO     timolo     Start Motion Tracking and TimeLapse Loop ... ctrl-c Exits
2021-04-25 11:43:51 INFO     timolo     daymode=True  MOTION_DOTS_ON=False 

2021-04-25 11:43:51 INFO     timolo     Sched TimeLapse  daymode=True  Timer=120 sec  ExitSec=0 0=Continuous
2021-04-25 11:43:51 INFO     timolo     Stop Motion Tracking PiVideoStream ...
Traceback (most recent call last):
  File "pi-timolo.py", line 2432, in <module>
    timolo()
  File "pi-timolo.py", line 2033, in timolo
    take_day_image(filename, TIMELAPSE_CAM_SLEEP_SEC)
  File "pi-timolo.py", line 1307, in take_day_image
    camera.capture(filename)
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 1418, in capture
    encoder.start(output)
  File "/usr/lib/python3/dist-packages/picamera/encoders.py", line 1126, in start
    super(PiCookedOneImageEncoder, self).start(output)
  File "/usr/lib/python3/dist-packages/picamera/encoders.py", line 375, in start
    self._open_output(output)
  File "/usr/lib/python3/dist-packages/picamera/encoders.py", line 331, in _open_output
    self.outputs[key] = mo.open_stream(output)
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 344, in open_stream
    stream = io.open(stream, 'wb' if output else 'rb', buffering)
PermissionError: [Errno 13] Permission denied: 'media/timelapse/tl-cam1-1223.png'
```

coder-cell, 2021-04-25:
@pjockey Check on the right path of the where the image is stored.

pjockey, 2021-04-25:
I have 9.5 GB free on the disk out of 16 GB. I went into the permissions on the folder and it stated "owner", changed to "anyone". I deleted all of the files out of those folders as well.

As I stated, it was working. How would these permissions change it in fact this is the problem? I did reboot the Pi and had the same results.

pageauc, 2021-04-25:
It is possible that if you have not run pi-timolo as another user like
running under sudo then problem could be a corrupt SD card  for some reason.

What permissions and ownership are on the problem files.  Eg In problem
folder do command  *ls -al*

/pi-timolo/media/timelapse $ ls -al
> total 40620
> drwxr-xr-x 2 pi pi    4096 Apr 23 10:58 .
> drwxr-xr-x 6 pi pi    4096 Apr 23 07:40 ..
> -rw-r--r-- 1 pi pi  130692 Apr 23 07:41 tl-box3-1000.jpg
> -rw-r--r-- 1 pi pi 1691808 Apr 23 07:45 tl-box3-1001.jpg
> -rw-r--r-- 1 pi pi 1702926 Apr 23 07:53 tl-box3-1002.jpg
> -rw-r--r-- 1 pi pi  811199 Apr 23 07:57 tl-box3-1003.jpg
> -rw-r--r-- 1 pi pi  810819 Apr 23 07:59 tl-box3-1004.jpg
> -rw-r--r-- 1 pi pi  818277 Apr 23 08:01 tl-box3-1005.jpg
> -rw-r--r-- 1 pi pi  932545 Apr 23 08:03 tl-box3-1006.jpg
> -rw-r--r-- 1 pi pi  265033 Apr 23 08:05 tl-box3-1007.jpg
>

Files should be owned by pi with write permissions for pi owner.  group and
global (world) get read only

closed 2021-05-18 by pageauc