pageauc/pi-timolo #121: IMAGE_USE_VIDEOPORT config is not used

npaibrooklyn, 2021-04-29:
use_video_port is hardcoded to True: https://github.com/pageauc/pi-timolo/blob/master/source/pi-timolo.py#L572

Consider removing this config variable or using it

pageauc, 2021-04-29:
Made change a while back and did not remove in case I wanted to revert
back. There are other things in the code
that probably need looking at but sometimes you create more problems when
tinkering with code that is working.  There are certainly areas for
optimization.
I normally use pylint and other tools to check code.  I am aware of other
issues as well.  This project
was originally done as a personal project that grew from two of my older
projects   rpi-timelapse with low light and motion tracking
I combined these and it grew from there.

Just curious but seems like you are doing some detailed audit and review of
my code.  I normally use  pylint and am aware of the miscellaneous issues
like unused variables and naming issues.  eg there are a mix of snake_case
and caMel case..  Other projects I see also have similar issues and it is
not a critical issue since this is simply a personal, shared project and
not production.  I am just a retired dude who codes as a hobby.

Your contributions  are appreciated..
Claude ...

npaibrooklyn, 2021-04-29:
Ah I understand now. Sorry I misunderstood it to be an active community project and I was trying to help maintain as and when I found issues.

pageauc, 2021-04-29:
I have implemented your change.  And I do appreciate contributions and
feedback.

The video port idea was included with a push by another user.  After
testing I later found it was not really required so I changed code but left
the config.py constant and it later got added with the config.py/ plugin
constant variable check dictionary audit routine.

Was just wondering why out of 2400 odd lines of code someone would report a
constant that was not being used and not creating any issue other than a
user trying to use the variable and nothing significant happening. An
average person would not discover this without an auditing tool. It would
be like a code comment with a not critical typo.

I normally study other users code to analyse the underlying algorithm(s),
copy a snippet to adapt for myself, Etc.  During these reads, I often find
useless or inactive logic, constants, data structures, branches Etc.  Lots
of projects have commented out code lines, blocks, functions, Etc that
serve no real purpose other than a reminder of code history, inactive
feature, Etc.. The code still runs just fine with or without these.

That is a complement to you
Claude ...

npaibrooklyn, 2021-04-29:
Thank you @pageauc ! I totally understand. Given I am a professional software engineer, I was in "better the repo and contribute" mode haha. I really appreciate this repo - it helped me get started on my fun personal project!

pageauc, 2021-04-29:
Covid Lockdown makes you appreciate human contact.  You just peaked my
curiosity, that is all.

  Thanks for the reply and Stay Safe.  Claude ...

npaibrooklyn, 2021-04-29:
@pageauc I completely understand - hopefully covid will be conquered soon! Do you have a linkedin account - wanted to add you as a friend there

pageauc, 2021-04-30:
I am retired.  Just use GitHub, YouTube and Twitter.  My wife uses Facebook
and misc messaging apps.  My US IPhone is only used for emergencies, no
data, just prepaid.  Besides Canadian data rates are probably the highest
in the world for no other reason but limited competition and greed.  I Just
buy a data pack for my wife's phone when we are travelling.
Regards
Claude ...

closed 2021-05-18 by pageauc