pageauc/pi-timolo #123: Low light video

konpaul, 2022-01-14:
Hey,
is there a possibility to save a low light video on a motion event? Or is that not possible due to the long shutter time?
I am using a IR Cut Camera.
Thanks for the help.
Cheers Paul

pageauc, 2022-01-14:
Paul

I have a RPI IR camera but I don't have it with me now.  I have used it
with Pi-Timolo but never got good results but then again I did not have an
IR light source.
Low light mode has too long an exposure to use motion tracking.  Normally
it would be best to have a motion activated security light near the camera,
then when the motion light comes on the camera would need some time to
track motion and trigger a video, photo, photo sequence Etc.

I have also played around with IR motion sensors connected to GPIO but
found them a little finicky and decided to use regular motion light not
attached to RPI.
Hope this answers your question
Claude ..

pageauc, 2022-01-15:
You might want to check out the NOIR camera python program below.

https://github.com/craigthomas/RPiDayNightCamera

pageauc, 2022-01-15:
Another python NOIR camera project.  See Source version 1.1.2 for latest
code.

https://github.com/DaProfessor/RPi_MoCap

konpaul, 2022-01-18:
thanks for your quick and detailed response! I will have a look at your suggestions

pageauc, 2025-02-23:
See NEW pi-timolo2 libcamera picamera2 at https://github.com/pageauc/pi-timolo2
I have not tested with noir camera yet

closed 2025-02-23 by pageauc