pageauc/pi-timolo #125: Bullseye default python version

MS-Stamp, 2022-02-23:
The latest Raspian build (Bullseye, Debian 11) only comes with python 3, so you have to manually install python 2 and then change the default version to get the webserver running. I used the following after a little searching on the web, but it would be good to either update the webserver for python 3 or document the workaround to support python 2.

sudo apt install python2
sudo apt install python-is-python2

pageauc, 2022-02-23:
Code is compatible with python3 but I have not run under Bullesye.  Not
sure if the PiCamera python library will work since
I believe they removed it from bullseye.  Let me know if it works with
bullseye

You can launch scripts under python 3 by manually changing the top shebang
to point to python3
Also webserver3.py is compatible with python3.  You can copy it over the
python.py and menubox should work

Let me know how you are doing

hughc, 2022-03-21:
I haven't had any luck with this under bullseye / python3. I'm new to the script, but got it working without issues a few months back. When I returned, I started afresh with a new SD image, which I assume included the OS update.  

```
hugh@zerotwo:~/pi-timolo $ ./pi-timolo.py
-------------------------------------------------------
pi-timolo.py ver 12.52  written by Claude Pageau
-------------------------------------------------------
Loading Wait ....
WARN  : Could Not Import pyexiv2. Required for Saving Image EXIF meta data
        If Running under python3 then Install pyexiv2 library for python3 per
        cd ~/pi-timolo
        ./install-py3exiv2.sh

Please Review Warnings  Wait 10 sec ...
Loading Wait ....
Traceback (most recent call last):
  File "/home/hugh/pi-timolo/./pi-timolo.py", line 395, in <module>
    from picamera import PiCamera
  File "/usr/lib/python3/dist-packages/picamera/__init__.py", line 72, in <module>
    from picamera.exc import (
  File "/usr/lib/python3/dist-packages/picamera/exc.py", line 41, in <module>
    import picamera.mmal as mmal
  File "/usr/lib/python3/dist-packages/picamera/mmal.py", line 49, in <module>
    _lib = ct.CDLL('libmmal.so')
  File "/usr/lib/python3.9/ctypes/__init__.py", line 374, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: libmmal.so: cannot open shared object file: No such file or directory
```

libmmal.so is present:

```
hugh@zerotwo:~/opencv3-setup $ ls /opt/vc/lib/libm*
/opt/vc/lib/libmmal_components.so  /opt/vc/lib/libmmal.so       /opt/vc/lib/libmmal_vc_client.so
/opt/vc/lib/libmmal_core.so        /opt/vc/lib/libmmal_util.so
```

I have enabled the legacy camera stack via `rpi-config`

I'm running on a Pi Zero 2 W. The github readme for python 3 suggests you need to build OpenCV, but when I download and run your build / install script, I get this:

```
./cv3-install-menu.sh ver 3.82    written by Claude Pageau

Internet Check OpenCV version 4.1.2  Wait ...

STATUS
Current Installed python OpenCV version is 4.5.1
Variable OPENCV_VER=4.1.2 Is a Valid OpenCV Version

UPGRADE
python OpenCV Version From 4.5.1
                       To  4.1.2
```

Which suggests a manual build is unnecessary?

pageauc, 2022-03-22:
There should be a setting using *sudo raspi-config* under   *Interface
Options*, called *I1 Legacy Camera Enable/disable legacy camera support*.
Try enabling this option then try to reinstall/rerun pi-timolo.  I Have not
tried this but you could test
to see if it will support the legacy picamera python library.  Let me know
if this works.

There should be not opencv issues and code can run under python2 or 3

Claude ....

hughc, 2022-03-22:
Sorry, I did not accurately explain that I had already enabled legacy camera support (I mentioned `rp-config`, rather than `raspi-config`).

The result remains the same, still seeing the libmmal error as previously quoted. This, after reinstalling after double-checking that camera support was enabled.

pageauc, 2022-03-22:
Thanks for the update

Until the new Bullseye and later picamera python library is released and
the pi-timilo code changed to work with that library then the only
alternative is to use raspberry pi OS prior to Bullseye
You can thank the raspberry pi OS team for the No Pain, No Gain attitude

pageauc, 2025-02-12:
I am making progress on updating pi-timolo to picamera2 libcamera. New repo will be called pi-timolo2 but it will be a while since I am still updating code and testing. Partially works at this time. In developing pi-timolo2, I created a test program called lolight. This uses a different low light algorithm.

https://github.com/pageauc/lolight

You can install a copy for testing per Readme.md. Let me know any issues. Should run with no extra dependencies.
In the mean time pi-timolo2 is still in progress.
Claude

transcendentphoto, 2025-02-16:
This is really great news, I admire your work and commitment. Thank you for the new version. I installed it, checked it - it works! Now I only miss the ability to change the resolution, but I know that this and other options will be available in the pi-timolo2 version. Congratulations, I'm waiting. Thank you.

pageauc, 2025-02-16:
I have successfully tested beta ver of timolo2.py.  Now working on cleaning
up unneeded files and modifying others. Using supervisor to manage running
webserver.py and timolo2.py as background tasks.
Algorithm for low light changed and looks good but needs more testing.
Code tested on Buster and Bookworm 64bit.  Runs good on both.
Install script is changed but needs testing from github.  Readme's and
general documentation will come later.  Currently pointing to wiki on
previous pi-timolo

timolo2.py is leaned up and is approx 500 lines shorter after code removals
and additions

Hope to post on github in a day or so.

pageauc, 2025-02-23:
See NEW pi-timolo2 libcamera picamera2 at https://github.com/pageauc/pi-timolo2

closed 2025-02-23 by pageauc