pageauc/pi-timolo #126: Update to Picamera2 for Raspbian Bullseye?

thjkral, 2022-03-15:
Do you plan to update the library for users that use the new Bullseye OS? Unfortunately, the integration of the camera with Python was not part of the release and makes a lot of projects incompatible. When Picamera2 is released, will there be a version of PI-TIMOLO that uses this new library? 

Switching back to the previous Buster version of Raspbian is not a long term solution for me and the other modules I run on my RasPi, I'm afraid.

pageauc, 2022-03-15:
Waiting for the full release of PiCamera2 that is not syntactically
compatible with the original PiCamera python library.  Will need a rewrite
of camera portions of Pi-Timolo.
I have discussed issues with PiCamera2 developers and was not pleased about
using a whole new syntax so not sure if I will create a new version or
integrate with existing.
Still on the fence at this time.
Claude ...

thjkral, 2022-03-17:
Thank you for the quick response. So Pi-Timolo will be available, in some shape or form, for the new OS in the future?

PeterIP4, 2022-11-25:
Hi. First I’d like to say what a wonderful idea.
I am trying to get this up and running but have had little success. It seems as if it Doesn’t yet work with later Raspian release, is that correct and if so which is the version I should use?
Regards Peter

pageauc, 2022-11-25:
It will run under Bullseye and anything earlier.  Not 64bit..  Just enable
Legacy picamera setting using sudo raspi-config under interfaces or if on
desktop select the menu pick for configuration settings.

Let me know how it goes.
Claude ...

PeterIP4, 2022-11-28:
Hi.

Some progress has been made by building from scratch with new copy of
Bullseye and the test works.

I still have a lot to do to work out how it all works and to decide if I
need py3exif2 and plug ins but it’s looking good.

Thanks for your help.

Regards
Peter

pageauc, 2022-11-28:
I started working on a test motion tracking program that uses picamera2 threaded streaming.  Works OK.  My sample code auto detects picamera and picamera2 configs as well as being able to access webcam and RTSP streams.  Everything works on this sample code.  Have not posted to Github.  This was mostly for updating speed camera but Timolo uses motion tracking as well.
I do plan at some time to rewrite new version for picamera2.  The existing one will be OK for backward compatibility.
exif routine is used to save and restore exif metadata that is lost when adding text to images.

Just to be clear are you just running existing timolo in Bullseye using the raspi-config, Interfaces menu pick that enables Legacy PiCamera.? or coding something.
Claude ...

rudolphthecat, 2024-12-28:
Are there any plans on supporting Bookworm?

pageauc, 2024-12-29:
Needs to be converted to libcam picamera 2 python library. Was looking at
leaving the original and creating an updated libcamera version. I have some
stream code for motion tracking, but was looking at lowlight code
compatible with libcam. Life gets in the way so I kind of put it on the
back burner since my timelapse cams are still bullseye on the original
picamera library.
Claude ...

pageauc, 2025-02-12:
I am making progress on updating pi-timolo.  New repo will be called pi-timolo2 but it will be a while since I am still updating code and testing. Partially works at this time.  In developing pi-timolo2, I created a test program called lolight.  This uses a different low light algorithm.

https://github.com/pageauc/lolight

You can install a copy for testing per Readme.md.  Let me know any issues. Should run with no extra dependencies.
In the mean time pi-timolo2 is still in progress.
Claude

pageauc, 2025-02-17:
Repo is now available.  Code is good enough but needs more testing. Has curl script for install.

https://github.com/pageauc/pi-timolo2

Let me know your comments.  You can share with others since I am not much of a social media guyt\.

Issues welcome.

Claude .....

pageauc, 2025-02-18:
Fixed up a few install bugs.  Should be OK now.  Use menubox.sh  UPGRADE menu pick to be sure.  Issue was with supervisor files.

Regards Claude ...

pageauc, 2025-02-23:
See NEW pi-timolo2 libcamera picamera2 at https://github.com/pageauc/pi-timolo2

closed 2025-02-23 by pageauc