pageauc/pi-timolo #129: pi-timolo exists due to error

GWAM, 2022-04-20:
Hi,
This has been working well for me, but stopped today, not sure why.
When I run ./pi-timolo I get:
```
-------------------------------------------------------
pi-timolo.py ver 12.52  written by Claude Pageau
-------------------------------------------------------
Loading Wait ....
2022-04-20 13:58:10 ERROR    <module>   Pi Camera Module Not Found supported=1 detected=1, libcamera interfaces=0
2022-04-20 13:58:10 ERROR    <module>   if supported=0 Enable Camera using command sudo raspi-config
2022-04-20 13:58:10 ERROR    <module>   if detected=0 Check Pi Camera Module is Installed Correctly
2022-04-20 13:58:10 ERROR    <module>   Exiting pi-timolo.py Due to Error
```

However, the camera is installed properly, and when I run
`sudo raspistill -o test.jpg`

An image is captured correctly with no problem, the camera responds
Earlier today I got for the first time the following error:
```
WARN  : Could Not Import pyexiv2. Required for Saving Image EXIF meta data
        If Running under python3 then Install pyexiv2 library for python3 per
        cd ~/pi-timolo
        ./install-py3exiv2.sh
```
So installed py3exiv2 and now only left with the error about the camera

Any ideas?
Steps I've taken so far - I reinstalled raspibian from scratch, then reinstalled pi-timolo but not solved

Thanks!

pageauc, 2022-04-20:
The python PiCamera python module is not supported on Bullseye.  Please
confirm if you are using legacy Buster Image which is the most recent image
that the python picamera module is supported on.  If you are running
Bullseye then install a Buster Legacy image then install pi-timolo.
Currently the Raspberry Pi foundation is working on the picamera2 python
library that will require a major rewrite.  It is just a demo at this point.
Let me know..

GWAM, 2022-04-21:
Good Morning Claude and thank you so much for the quick response, not taken for granted, I know you do this for fun so truly appreciate!
I verified and you are right, I am running bullseye, I will try to downgrade back to a buster image and re-install pi-timolo and see what happens, I'll report back here soon.

Thanks

pageauc, 2022-04-21:
I will look at adding some code to check for OS version greater than or
equal to Bullseye and display a meaningful error message.

GWAM, 2022-04-21:
Issue close.
As Claude mentioned, my upgrade to bullseye made picam not supported anymore, I downgraded back to buster, re-installed pi-timolo and now everything is back in order working, thanks!

closed 2022-04-21 by GWAM