pageauc/pi-timolo #133: "pi" is no longer the standard user name on Raspberry PIs

askpatrickw, 2023-05-18:
Lots of the code assumes /home/pi is the path and pi is user.
It is no longer the given that "pi" is the user name everyone is using.

Some alternative approaches to take:

- .sh: use "~" or $HOME
- .py: use os.getenv('HOME')

I am not sure if those will also work when running as a service.

pageauc, 2025-02-23:
See NEW pi-timolo2 libcamera picamera2 at https://github.com/pageauc/pi-timolo2

I will look at this issue for new picamera2 release.  Will add as issue on pi-timolo2

closed 2025-02-23 by pageauc

pageauc, 2025-02-25:
I have modified the NEW pi-timolo2 libcamera picamera2 at https://github.com/pageauc/pi-timolo2
It will run under a different RPI login account.  This is used for timolo2-cam.sh and timolo2-web.sh for launching the related python scripts in the background using supervisor.  menuboxsh also works under a different user.  You will have to manually modify makevideo.conf although this could also be updated using sed.

If you have time I would appreciate any issues you encounter running my pi-timolo2 project.
Thanks Claude ....